ruuvi/ruuvi.gateway_nrf.c #2: UART TX buffer exceeded if Filter is off

theBASTI0N, 2019-12-20:
If the Ruuvi filter is removed and all BLE data is sent over UART the buffer gets overloaded and a reset is caused.
```bash
 <error> uart: Failed sending data to UART: 4
[00:00:14.682,006] <error> dongle: err: 4 - NRF_ERROR_NO_MEM,line: 51, file: ./src/uart.c

[00:00:00.000,549] <info> ble: interval: 10400
[00:00:00.000,549] <info> ble: channel_mask:
[00:00:00.000,549] <info> ble:  00 00 00 00 00         |.....   
[00:00:00.000,549] <info> ble:  00 00 00 00 E0         |.....   
[00:00:00.000,579] <debug> ble: Setting scanning channel: ch: 37, b: 4, bit: 5, mask: 0xDF
[00:00:00.000,579] <info> dongle: Ruuvi Dongle started.
[00:00:00.000,579] <debug> ble: Scan start
[00:00:00.000,671] <debug> ble_scan: Scanning
```bash

When UART_TX_BUF_SIZE is set to  512 the error is no longer seen.

This was error was seen with the default RUUVI_SCAN_PERIOD left as default.

ojousima, 2019-12-20:
@jsaari ^

jsaari, 2019-12-20:
How many BLE devices (or advertisements per second) you need to trigger this?

theBASTI0N, 2019-12-20:
I have about 20 to 30 BLE devices in range. I would assume that if they are all Ruuvi tags it would also occur. 

This is also being run from a Ruuvitag so the gateway with a better antenna could see this occur easier as it would be able to see more BLE Devices.

jsaari, 2019-12-20:
I think less than 10 devices with 100 ms interval was enough.

theBASTI0N, 2019-12-20:
Tested the new commit, haven't been able to see this issue again.

closed 2019-12-20 by theBASTI0N